FreeRTOS/FreeRTOS #1272: Update CORTEX R4 demo script to support non-Windows platforms

mryall-mawson, 2024-09-26:
**Is your feature request related to a problem? Please describe.**
The Cortex R4 demo code for TMS570 and RM48 has a batch file to copy some files around. This batch file only works on Windows.

**Describe the solution you'd like**
I've converted the batch file into a shell script which will work equivalently for Linux and MacOS dev environments.

**Additional context**
I'll open a PR with this enhancement shortly for you to review.

mryall-mawson, 2024-09-26:
PR is open: #1273 

It includes a small bug fix to the batch file as well, which was checking for a non-existent directory.

jasonpcarroll, 2024-09-26:
Hi @mryall-mawson,
I will review your PR.
Best,
Jason Carroll

mryall-mawson, 2024-09-27:
Thanks Jason. I have updated the PR to replace the scripts with symlinks, as suggested. This has been tested locally, and works for me with CCS12.

Once the PR is merged, we will also need to update [this section of the documentation](https://freertos.org/Documentation/02-Kernel/03-Supported-devices/04-Demos/Texas-Instruments/Free_RTOS_for_TI_RM48_and_TMS570#preparing-the-code-composer-studio-eclipse-project-directory). Where can I find the source for that?

closed 2024-10-10 by DakshitBabbar

rawalexe, 2024-10-10:
Thank you for your contribution, @DakshitBabbar  was able to merge your PR and also update the website.  Hence we closed the issue. Please go through the changes and let us know if you have any further queries