FreeRTOS/FreeRTOS #1274: [BUG] Kernel unit tests "Failed to CreateArtifact" with 409 Conflict on GitHub Actions

mryall-mawson, 2024-09-28:
**Describe the bug**
The kernel unit tests are failing on GitHub Actions ([example](https://github.com/FreeRTOS/FreeRTOS/actions/runs/11064752601/job/30781381289?pr=1273#step:7:20), [another example](https://github.com/FreeRTOS/FreeRTOS/actions/runs/11031358789/job/30637922006#step:7:20)) with the following error:

```
Failed to CreateArtifact: Received non-retryable error: Failed request: (409) Conflict: an artifact with this name already exists on the workflow run
```

This occurs in the "Archive code coverage data" step, with the settings:

```
with:
    name: coverage-data
    path: FreeRTOS/Test/CMock/build/cmock_test*
```

**Investigation**
Looking at the GitHub Actions config, commit 9245b4aaac0e2cd1fadff4fe701dcc99030d420c was made two weeks ago, which upgraded `upload-artifact` task from v2 to v4.

There was a [published breaking change in upload-artifact v4](https://github.com/actions/upload-artifact?tab=readme-ov-file#breaking-changes):

> Due to how Artifacts are created in this new version, it is no longer possible to upload to the same named Artifact multiple times. You must either split the uploads into multiple Artifacts with different names, or only upload once. Otherwise you will encounter an error.

This affects the coverage artifacts in the unit test action config, which use the same artifact name across jobs.

(Personally I think this should be fixed in the GH side, but it looks like the GH Actions devs are [declining requests](https://github.com/actions/upload-artifact/issues/481#issuecomment-1862899981) to restore the previous functionality. 😞)

**To Reproduce**
- See [all the builds](https://github.com/FreeRTOS/FreeRTOS/actions) failing on `main` for the last two weeks.

**Additional context**
I'll open a PR shortly to deduplicate the artifact names in the kernel unit tests action config.

mryall-mawson, 2024-09-28:
PR is open: #1275

aggarg, 2024-09-30:
Thank you for doing the detailed investigation and fixing the issue.

closed 2024-09-30 by aggarg