FreeRTOS/FreeRTOS #1282: Missing volatile qualifier

bluewww, 2024-10-16:
The  `readb` and `writeb` operations both are missing volatile qualifiers

https://github.com/FreeRTOS/FreeRTOS/blob/0e8d2bf66cd13353e5fe08997196895d9257c445/FreeRTOS/Demo/RISC-V-Qemu-virt_GCC/ns16550.c#L55-L63

For example this code 

https://github.com/FreeRTOS/FreeRTOS/blob/b3e6cff1f2392ceb261ac6e4b61aede2edc1ecf7/FreeRTOS/Demo/RISC-V-Qemu-virt_GCC/ns16550.c#L65-L74

compiles to

```
0000000080000390 <vOutNS16550>:
    80000390:   6118                    ld      a4,0(a0)
    80000392:   00574783                lbu     a5,5(a4)
    80000396:   0207f793                andi    a5,a5,32
    8000039a:   c781                    beqz    a5,800003a2 <vOutNS16550+0x12>
    8000039c:   00b70023                sb      a1,0(a4)
    800003a0:   8082                    ret
    800003a2:   a001                    j       800003a2 <vOutNS16550+0x12>
``` 

under gcc 13.2.0 and `-O2 -g` which is clearly broken (infinite loop after one access).

aggarg, 2024-10-16:
The above works for me. Here is the assembly before the change-
```asm
Disassembly of section .text.vOutNS16550:

00000000 <vOutNS16550>:
   0:   4118                    lw      a4,0(a0)
   2:   00574783                lbu     a5,5(a4)
   6:   0207f793                andi    a5,a5,32
   a:   e391                    bnez    a5,e <.L5>

0000000c <.L4>:
   c:   a001                    j       c <.L4>

0000000e <.L5>:
   e:   00b70023                sb      a1,0(a4)
  12:   8082                    ret
```

And here is after the change-
```asm
Disassembly of section .text.vOutNS16550:

00000000 <vOutNS16550>:
   0:   411c                    lw      a5,0(a0)
   2:   1141                    addi    sp,sp,-16
   4:   c63e                    sw      a5,12(sp)

00000006 <.L2>:
   6:   47b2                    lw      a5,12(sp)
   8:   0057c783                lbu     a5,5(a5)
   c:   0207f793                andi    a5,a5,32
  10:   dbfd                    beqz    a5,6 <.L2>
  12:   47b2                    lw      a5,12(sp)
  14:   00b78023                sb      a1,0(a5)
  18:   0141                    addi    sp,sp,16
  1a:   8082                    ret
```

bluewww, 2024-10-16:
I personally favor  `return *( (volatile uint8_t *) addr );`  and `*( (volatile uint8_t *) addr ) = b;`  in `readb` and `writeb` respectively as this is how I usually do it as the memory access is volatile not the address itself.

aggarg, 2024-10-16:
Fair enough - I updated the PR as per your suggestion.

bluewww, 2024-10-16:
That looks correct to me 👍

closed 2024-10-17 by aggarg