FreeRTOS/FreeRTOS #1321: [BUG]

neilklingensmith, 2025-01-20:
**Describe the bug**
FreeRTOS kernel hangs when sending to a full queue from an ISR if that queue is part of a select set.

**Target**
- Development board: Microchip SAMD20
- Instruction Set Architecture: ARM Cortex-M0+
- IDE and version: gcc
- Toolchain and version: arm-none-eabi-gcc version 13.2.1 20231009 (15:13.2.rel1-2) 

**Host**
- Host OS: Ubuntu
- Version: 24.04LTS

**To Reproduce**
1. Create a pair of queues with `xQueueCreate()`
2. Create a queue set with `xQueueCreateSet()`
3. Add both queues to the set with `xQueueAddToSet()`
4. Select on the queue set by calling `xQueueSelectFromSet()`
5. Send lots of data to one of the queues from an ISR by calling `xQueueSendFromISR()` to fill up one queue without giving the userspace code an opportunity to remove data.
6. `prvNotifyQueueSetContainer()` hangs on line 3320 of queue.c (offending line of code below). This appears to be an unnecessary assertion that checks if there is enough space in the queue to add another element. The following line of code performs the same check without an assert in an `if()...else` block. The `if()...else` block seems to be sufficient to check that there is enough space in the queue set container. If the container is full, the else block should set `xReturn = pdFALSE` and return an error to the user without hanging.

        configASSERT( pxQueueSetContainer->uxMessagesWaiting < pxQueueSetContainer->uxLength );

To fix, remove `configASSERT()` check and allow the check to be done by the `if()...else` block that immediately follows.

**Expected behavior**
If the queue set is full, `xQueueSendFromISR()` should return an error, not hang.

aggarg, 2025-01-21:
Do you have the callstack when you see this hang? `prvNotifyQueueSetContainer` is an internal function and this condition is ensured to be true before calling this function.

neilklingensmith, 2025-01-21:
Right, I see there is a check on line 1198 (of my file) of queue.c that checks

    if( ( pxQueue->uxMessagesWaiting < pxQueue->uxLength ) || ( xCopyPosition == queueOVERWRITE ) )

It looks like this check is made in `xQueueGenericSendFromISR()`, which then calls `prvCopyDataToQueue()` on line 1210 of my file. `prvCopyDataToQueue()` increments `pxQueue->uxMessagesWaiting`. If the queue is almost full, the condition checked on line 1198 will no longer be true. When `prvNotifyQueueSetContainer()` is called a few lines below, the condition is no longer true, and the assert hangs.

Stack trace below:

    prvNotifyQueueSetContainer()  <-- Top of stack
    xQueueGenericSendFromISR()
    <Interrupt Service Routine>

aggarg, 2025-01-22:
The check in the `xQueueGenericSendFromISR` is on the actual queue which checks that there is space in the queue. The check in the `prvNotifyQueueSetContainer` is on the queue set which checks that there is space in the queue set. I think this assert is useful because there is no other way to inform the application about the lost event. Is it possible that you are not setting the length of the queue set large enough as described [here](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/queue.h#L1663-L1677):
```
uxEventQueueLength should be set to the total sum of the length of the queues added to the set.
```

neilklingensmith, 2025-01-22:
I understand. Seems like it would make more sense to simply return an error code to the caller and let the caller deal with the condition by logging the error or just moving on. FreeRTOS does love asserts though, so maybe that's just a philosophical difference.

closed 2025-01-22 by neilklingensmith