FreeRTOS/FreeRTOS #1330: How can I use FreeRTOS to discover and connect to Greengrass core device

BasemTarek, 2025-02-23:
I am wondering why did freertos no longer have [aws_greengrass_discovery](https://github.com/aws/amazon-freertos/tree/main/libraries/freertos_plus/aws/greengrass) library!?

How can I use FreeRTOS to discover and connect to Greengrass core device?

BasemTarek, 2025-02-25:
Thanks @aggarg for your response

May you please provide a more descriptive steps about doing this: 

> You can use the [coreHTTP](https://github.com/FreeRTOS/coreHTTP) library to make the HTTP call and [coreJSON](https://github.com/FreeRTOS/coreJSON) to parse the JSON response.

Thanks,

aggarg, 2025-02-26:
> How to use the Connectivity Information? - (CA, Endpoint ..)

You should be able to use the same X.509 certificate that you are using to connect to AWS IoT Core. You need to provide GreenGrass permissions in your policy document. [This page](https://docs.aws.amazon.com/greengrass/v2/developerguide/greengrass-discover-api.html) provides all the details. 

> May you please provide a more descriptive steps about doing this:

If you read the `aws_greengrass_discovery` file that you linked, it does 2 things:
1. It makes an HTTP call to get a JSON file.
2. It then parses the JSON file and extracts GG Core information from that. 

You can use coreHTTP for the 1 and coreJSON for the 2. If you looking for examples on how to use these libraries, you can look [here](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/coreHTTP_Windows_Simulator) and [here](https://github.com/FreeRTOS/coreJSON/blob/main/test/unit-test/core_json_utest.c). Before you do that, I'd suggest to look if the example that @xuelix shared serves your purpose.

BasemTarek, 2025-02-26:
Thanks @xuelix and @aggarg for your response again,

I tried to run the suggested example and I got the following response:

`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[WARN] [DEMO] [greengrass_demo_local_auth.c:375] Connection to the broker failed. Retrying connection after 184 ms backoff.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[WARN] [DEMO] [greengrass_demo_local_auth.c:375] Connection to the broker failed. Retrying connection after 25 ms backoff.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[WARN] [DEMO] [greengrass_demo_local_auth.c:375] Connection to the broker failed. Retrying connection after 853 ms backoff.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[WARN] [DEMO] [greengrass_demo_local_auth.c:375] Connection to the broker failed. Retrying connection after 250 ms backoff.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[WARN] [DEMO] [greengrass_demo_local_auth.c:375] Connection to the broker failed. Retrying connection after 4232 ms backoff.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:339] Establishing a TLS session to 192.168.100.55:8883.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:275] SSL_connect failed to perform TLS handshake.`
`[ERROR] [Transport_OpenSSL_Sockets] [openssl_posix.c:678] Failed to establish a TLS connection.`
`[ERROR] [DEMO] [greengrass_demo_local_auth.c:370] Connection to the broker failed, all attempts exhausted.`
`[ERROR] [DEMO] [greengrass_demo_local_auth.c:895] Failed to connect to MQTT broker 192.168.100.55.`
`[INFO] [DEMO] [greengrass_demo_local_auth.c:914] Short delay before starting the next iteration....`
`

Actually I'm confused about which device certificate should I use to enable the client to connect to the greengrass, as the example [says](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/greengrass/greengrass_demo_local_auth#:~:text=Set%20this%20to%20the%20client%20cert%20downloaded%20from%20IoT%20core%20when%20setting%20up%20the%20Thing.) 

> `CLIENT_CERT_PATH`

> Set this to the client cert downloaded from IoT core when setting up the Thing.

and [also says](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/greengrass/greengrass_demo_local_auth#:~:text=Create%20a%20new%20thing%20and%20link%20it%20with%20this%20new%20certificate%20thing_cert.pem%20and%20set%20the%20value%20of%20the%20macro%20THING_NAME%20in%20demo_config.h%20file%20to%20the%20name%20of%20this%20new%20thing) 

> Create a new thing and link it with this new certificate thing_cert.pem and set the value of the macro THING_NAME in demo_config.h file to the name of this new thing

Is that mean that I should set `CLIENT_CERT_PATH` to the client cert downloaded from IoT core when setting up the Thing while at the same time link the generated certificate to the thing on the cloud?

Best regards,

BasemTarek, 2025-02-27:
I configured the aws.greengrass.clientdevices.Auth component with the generated CA path on the greengrass core device, and registered the generated client certificate and private key with the newly created thing on the cloud as described in the demo, still get the error.

When I run the following command:
`openssl s_client -connect 192.168.4.63:8883 -CAfile /home/basem/Documents/Greengrass_local_auth_demo/ca.pem -cert /home/basem/Documents/Greengrass_local_auth_demo/thing_cert.pem -key /home/basem/Documents/Greengrass_local_auth_demo/thing_private.key`

I got this response:

```
CONNECTED(00000003)
Can't use SSL_get_servername
depth=1 C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA
verify error:num=19:self-signed certificate in certificate chain
verify return:1
depth=1 C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA
verify return:1
depth=0 C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = aws.greengrass.clientdevices.mqtt.Moquette
verify return:1

Certificate chain
 0 s:C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = aws.greengrass.clientdevices.mqtt.Moquette
   i:C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA
   a:PKEY: rsaEncryption, 4096 (bit); sigalg: RSA-SHA256
   v:NotBefore: Feb 27 08:16:15 2025 GMT; NotAfter: Mar  6 08:16:15 2025 GMT
 1 s:C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA
   i:C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Feb 26 22:33:13 2025 GMT; NotAfter: Feb 25 22:33:13 2030 GMT

Server certificate
-----BEGIN CERTIFICATE-----
.....
-----END CERTIFICATE-----

subject=C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = aws.greengrass.clientdevices.mqtt.Moquette
issuer=C = US, O = Amazon.com Inc., OU = Amazon Web Services, ST = Washington, L = Seattle, CN = Greengrass Core CA

No client certificate CA names sent
Client Certificate Types: ECDSA sign, RSA sign, DSA sign
Requested Signature Algorithms: ECDSA+SHA256:ECDSA+SHA384:ECDSA+SHA512:Ed25519:Ed448:RSA-PSS+SHA256:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA-PSS+SHA256:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA+SHA256:RSA+SHA384:RSA+SHA512:DSA+SHA256:ECDSA+SHA224:RSA+SHA224:DSA+SHA224:ECDSA+SHA1:RSA+SHA1:DSA+SHA1
Shared Requested Signature Algorithms: ECDSA+SHA256:ECDSA+SHA384:ECDSA+SHA512:Ed25519:Ed448:RSA-PSS+SHA256:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA-PSS+SHA256:RSA-PSS+SHA384:RSA-PSS+SHA512:RSA+SHA256:RSA+SHA384:RSA+SHA512:DSA+SHA256:ECDSA+SHA224:RSA+SHA224:DSA+SHA224
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits

SSL handshake has read 8104 bytes and written 2698 bytes
Verification error: self-signed certificate in certificate chain

New, TLSv1.2, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 4096 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: 860DD1EDE1602BE80D0FB796D2F01492C2D9879C049BB3E82C61A726547D12A5
    Session-ID-ctx: 
    Master-Key: A44D07EFA643F1ACCE4D8D93BEEB274106D7F527F99CB06FFBEF23ECCF53EE61FCB9FC3511318277247E7CDC7A884B22
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 86400 (seconds)
    TLS session ticket:

.....

    Start Time: 1740649812
    Timeout   : 7200 (sec)
    Verify return code: 19 (self-signed certificate in certificate chain)
    Extended master secret: yes

```

BasemTarek, 2025-02-27:
**sudo tail -f /greengrass/v2/logs/greengrass.log**

```
2025-02-27T10:29:23.819Z [INFO] (nioEventLoopGroup-3-8) io.moquette.broker.metrics.MQTTMessageLogger: Channel Inactive. {}
2025-02-27T10:31:09.622Z [ERROR] (nioEventLoopGroup-3-1) io.moquette.broker.NewNettyMQTTHandler: Unexpected exception while processing MQTT message. Closing Netty channel. CId=null. {}
io.netty.handler.codec.DecoderException: javax.net.ssl.SSLHandshakeException: Received fatal alert: unknown_ca
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:500)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:290)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:444)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.timeout.IdleStateHandler.channelRead(IdleStateHandler.java:289)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.moquette.broker.metrics.BytesMetricsHandler.channelRead(BytesMetricsHandler.java:51)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: unknown_ca
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:130)
	at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:117)
	at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:370)
	at java.base/sun.security.ssl.Alert$AlertConsumer.consume(Alert.java:287)
	at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:209)
	at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:172)
	at java.base/sun.security.ssl.SSLEngineImpl.decode(SSLEngineImpl.java:736)
	at java.base/sun.security.ssl.SSLEngineImpl.readRecord(SSLEngineImpl.java:691)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:506)
	at java.base/sun.security.ssl.SSLEngineImpl.unwrap(SSLEngineImpl.java:482)
	at java.base/javax.net.ssl.SSLEngine.unwrap(SSLEngine.java:679)
	at io.netty.handler.ssl.SslHandler$SslEngineType$3.unwrap(SslHandler.java:310)
	at io.netty.handler.ssl.SslHandler.unwrap(SslHandler.java:1445)
	at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1338)
	at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1387)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:530)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:469)
	... 25 more
```

aggarg, 2025-03-04:
> Caused by: javax.net.ssl.SSLHandshakeException: Received fatal alert: unknown_ca

Did you register the CA with AWS IoT as described in the [README](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/greengrass/greengrass_demo_local_auth/README.md)?

BasemTarek, 2025-03-04:
Yes I did.

BasemTarek, 2025-03-05:
@xuelix helped me solve the issue [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/1936#issuecomment-2699323170), the problem was related to the certificate name.

I'll try the same way to enable the esp32/freertos to connect to the greengrass.

Thanks,

closed 2025-03-05 by BasemTarek