FreeRTOS/FreeRTOS #1333: [Feature Request] Allow the user to manage MPU regions in the Cortex M55 port

ErickReyesR, 2025-03-18:
The FreeRTOS MPU port for this processor takes full control of the entire MPU.

There are some scenarios where the user may want to manage the some MPU regions themselves. For example: I could not find a way to set a region caching attributes to anything other than "Non-Transient, Write-back, Read-Allocate and Write-Allocate." or "Non-cacheable".

The existing code supports hardware configurations with either 8 or 16 MPU regions, where the user has to define "configTOTAL_MPU_REGIONS" in their FreeRTOSConfig.h to match the hardware.

I have tested a simple code change where the user can define another symbol in their config file, named "configUSER_MANAGED_MPU_REGIONS". To keep things simple, I made this only valid when the hardware has 16 MPU regions. The FreeRTOS kernel retains control of the first 8 and leaves the other 8 to the user. This should also improve context switching time a little since only 8 regions are saved/restored.

Is there a chance this change can be accepted?

Below is a text version of the patch, which I can also send as a pull request:

```
diff --git a/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/port.c b/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/port.c
--- a/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/port.c
+++ b/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/port.c
@@ -177,7 +177,7 @@
 #define portMPU_ENABLE_BIT                    ( 1UL << 0UL )
 
 /* Expected value of the portMPU_TYPE register. */
-#define portEXPECTED_MPU_TYPE_VALUE           ( configTOTAL_MPU_REGIONS << 8UL )
+#define portEXPECTED_MPU_TYPE_VALUE           ( (configTOTAL_MPU_REGIONS + configUSER_MANAGED_MPU_REGIONS) << 8UL )
 /*-----------------------------------------------------------*/
 
 /**
diff --git a/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/portmacrocommon.h b/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/portmacrocommon.h
--- a/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/portmacrocommon.h
+++ b/FreeRTOS/Source/portable/GCC/ARM_CM55_NTZ/non_secure/portmacrocommon.h
@@ -139,6 +139,16 @@
     #define configTOTAL_MPU_REGIONS       ( 8UL )
 #endif
 
+/*
+ * By default, the kernel manages all MPU regions.
+ * However, on systems with 16 MPU regions, the user can decide
+ * to manage the last 8. The only valid values are 0 and 8.
+ */
+#ifndef configUSER_MANAGED_MPU_REGIONS
+#define configUSER_MANAGED_MPU_REGIONS (0UL)
+#endif
+
+
 /* MPU regions. */
     #define portPRIVILEGED_FLASH_REGION                   ( 0UL )
     #define portUNPRIVILEGED_FLASH_REGION                 ( 1UL )
```

My application uses the feature by defining these in the config file:
```
/* MPU configuration - leave 8 MPU regions under control of FreeRTOS. */
#define configTOTAL_MPU_REGIONS 8

/* Define user managed regions that we control so the total number of regions is 16.
 */
#define configUSER_MANAGED_MPU_REGIONS 8
```

aggarg, 2025-03-18:
Thank you for reporting this. Instead of adding a new configuration parameter, we can remove the following assert and if check: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM55/non_secure/port.c#L929-L932. This way you can set `configTOTAL_MPU_REGIONS` to 8 and then manage MPU regions 8-15 in the application. 

Additionally, we should look into if we can enable the application to set memory attributes for kernel managed MPU regions. This may require more work though.

PS - Would you please move this issue to the [FreeRTOS-Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main) repo.

ErickReyesR, 2025-03-18:
Thank you for the quick reply.

What you suggest would also solve my use case.

It looks like I'd need write access on these repositories to move the issue. Am I missing something?