FreeRTOS/FreeRTOS #1337: Changes to conditional statements due to changes in MbedTLS v3.6.3 specifications

KeitaKashima, 2025-04-14:
**Describe the bug**

I think it is not a bug, but it doesn't work when combined with MbedTLS v3.6.3 or later.

And the update of the MbedTLS is for a Vulnerability. Then, there are people who will face the same issue when combined with that version of Mbe.

The mbedtls_ssl_handshake() of MbedTLS function returns "False" when the "pNetworkCredentials->disableSni" is `pdTRUE` because the MbedTLS changed the specification.

**Target**
-  network_transport layer : https://github.com/FreeRTOS/FreeRTOS/blob/86d6312da0263d76749f97eeb9722b35b8e0fefb/FreeRTOS-Plus/Source/Application-Protocols/network_transport/transport_mbedtls.c#L461
- Development board: all
- Instruction Set Architecture: all

**Host**
- Host OS: all

**Expected behavior**

Run the code when `pNetworkCredentials->disableSni` is `pdTRUE`.

**Screenshots**

The `mbedtls_ssl_handshake()` returns the `False` when using MbedTLS v3.6.3 or later.

https://github.com/FreeRTOS/FreeRTOS/blob/86d6312da0263d76749f97eeb9722b35b8e0fefb/FreeRTOS-Plus/Source/Application-Protocols/network_transport/transport_mbedtls.c#L598

Because, the function requires to call the `mbedtls_ssl_set_hostname()` before calling mbedtls_ssl_handshake(), but the `mbedtls_ssl_set_hostname()` only is called when "pNetworkCredentials->disableSni" is `pdFALSE`.

[image]

https://github.com/FreeRTOS/FreeRTOS/blob/86d6312da0263d76749f97eeb9722b35b8e0fefb/FreeRTOS-Plus/Source/Application-Protocols/network_transport/transport_mbedtls.c#L461

[image]

* About MbetTLS updated

https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.3

[image]

kar-rahul-aws, 2025-04-14:
Hi @KeitaKashima 
Thank you for reporting the issue. We are checking with the team. In the meanwhile, would you like to propose a PR for solving the issue ?

KeitaKashima, 2025-04-16:
Hi @kar-rahul-aws 

> Hi [@KeitaKashima](https://github.com/KeitaKashima) Thank you for reporting the issue. We are checking with the team. In the meanwhile, would you like to propose a PR for solving the issue ?

Sorry for the late reply as I was out.
I don't have a fixing code but your code [Update network_transport files based on MbedTLSv3.6.3 #1340](https://github.com/FreeRTOS/FreeRTOS/pull/1340) is the same idea as my fix.

closed 2025-04-21 by kar-rahul-aws