FreeRTOS/FreeRTOS #1365: [BUG] Explicit casts to function pointer types with latest GCC compiler version

VanNamDinh, 2025-07-15:
**Describe the bug**
I tried latest GCC compiler version [GCC for Renesas 14.2.0.202505-GNURX], the build error happened with following files:
- mbedtls_pk_pkcs11.c
- transport_mbedtls_pkcs11.c
- UARTCommandConsole.c

The function pointers without explicit casting. This can cause compiler errors.
What we need to do is add explicit casts to function pointer types.

**Target**
- IDE and version: [e2studio 2025-04]
- Toolchain and version: [GCC for Renesas 14.2.0.202505-GNURX]

**Host**
- Host OS: [Windows 10]

**To Reproduce**

1. Fix transport_mbedtls_pkcs11.c as below:
```c
mbedtls_pk_info_t mbedtls_pkcs11_pk_ecdsa =
{
    .type               = MBEDTLS_PK_ECKEY,
    .name               = "PKCS#11",
    .get_bitlen         = (size_t (*)(mbedtls_pk_context *))p11_ecdsa_get_bitlen,
    .can_do             = p11_ecdsa_can_do,
    .verify_func        = p11_ecdsa_verify,
    .sign_func          = p11_ecdsa_sign,
    #if defined( MBEDTLS_ECDSA_C ) && defined( MBEDTLS_ECP_RESTARTABLE )
        .verify_rs_func = NULL,
        .sign_rs_func   = NULL,
    #endif /* MBEDTLS_ECDSA_C && MBEDTLS_ECP_RESTARTABLE */
    .decrypt_func       = NULL,
    .encrypt_func       = NULL,
    .check_pair_func    = (int (*)(mbedtls_pk_context *, mbedtls_pk_context *, int (*)(void *, unsigned char *, size_t),void *))p11_ecdsa_check_pair,
    .ctx_alloc_func     = p11_ecdsa_ctx_alloc,
    .ctx_free_func      = p11_ecdsa_ctx_free,
    #if defined( MBEDTLS_ECDSA_C ) && defined( MBEDTLS_ECP_RESTARTABLE )
        .rs_alloc_func  = NULL,
        .rs_free_func   = NULL,
    #endif /* MBEDTLS_ECDSA_C && MBEDTLS_ECP_RESTARTABLE */
    .debug_func         = (void (*)(mbedtls_pk_context *, mbedtls_pk_debug_item *))p11_ecdsa_debug,
};
```

```c
mbedtls_pk_info_t mbedtls_pkcs11_pk_rsa =
{
    .type               = MBEDTLS_PK_RSA,
    .name               = "PKCS#11",
    .get_bitlen         = (size_t (*)(mbedtls_pk_context *))p11_rsa_get_bitlen,
    .can_do             = p11_rsa_can_do,
    .verify_func        = p11_rsa_verify,
    .sign_func          = p11_rsa_sign,
    #if defined( MBEDTLS_ECDSA_C ) && defined( MBEDTLS_ECP_RESTARTABLE )
        .verify_rs_func = NULL,
        .sign_rs_func   = NULL,
    #endif /* MBEDTLS_ECDSA_C && MBEDTLS_ECP_RESTARTABLE */
    .decrypt_func       = NULL,
    .encrypt_func       = NULL,
    .check_pair_func    = (int (*)(mbedtls_pk_context *, mbedtls_pk_context *, int (*)(void *, unsigned char *, size_t),void *))p11_rsa_check_pair,
    .ctx_alloc_func     = p11_rsa_ctx_alloc,
    .ctx_free_func      = p11_rsa_ctx_free,
    #if defined( MBEDTLS_ECDSA_C ) && defined( MBEDTLS_ECP_RESTARTABLE )
        .rs_alloc_func  = NULL,
        .rs_free_func   = NULL,
    #endif /* MBEDTLS_ECDSA_C && MBEDTLS_ECP_RESTARTABLE */
    .debug_func         = (void (*)(mbedtls_pk_context *, mbedtls_pk_debug_item *))p11_rsa_debug,
};
```
2. In transport_mbedtls_pkcs11.c with tlsSetup() function, fix as below

```c
mbedtls_ssl_conf_rng( &( pTlsTransportParams->sslContext.config ),
                              (void *)generateRandomBytes,
                              &pTlsTransportParams->sslContext );
```

3. Fix L109 in UARTCommandConsole.c

```c
static void prvUARTCommandConsoleTask( void *pvParameters )
{
signed char cPrevChar = (signed char) NULL;
```

**Expected behavior**
Build passes without warning

**Screenshots**

**Additional context**

jasonpcarroll, 2025-07-15:
Hi @VanNamDinh,

jasonpcarroll, 2025-07-15:
Actually, it looks like you already have. This PR is to fix this correct https://github.com/FreeRTOS/FreeRTOS/pull/1366?