FreeRTOS/FreeRTOS #1367: [BUG] SMP Testing Issue on AARCH64: Individual Tests Pass but Combined Tests Fail

github-xiaodong, 2025-07-18:
**Describe the bug**

I’m porting FreeRTOS to a 4-core AARCH64 SMP system (based on the [CORTEX_A53_64-bit_UltraScale_MPSoC](https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/main/GCC/CORTEX_A53_64-bit_UltraScale_MPSoC) implementation. 

When running the test suite following the [ThirdParty/Template README](https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS/Demo/ThirdParty/Template/README.md), I observe:

✅ Individual tests pass when executed alone (e.g., configSTART_BLOCKING_QUEUE_TESTS or configSTART_GENERIC_QUEUE_TESTS).

❌ Combined tests fail when multiple test cases are selected together (e.g., configSTART_BLOCKING_QUEUE_TESTS+ configSTART_GENERIC_QUEUE_TESTS + configSTART_ABORT_DELAY_TESTS).

**Environment**

FreeRTOS Source: [FreeRTOS-Kernel main branch](https://github.com/FreeRTOS/FreeRTOS-Kernel) (commit: [latest]).

Target: 4-core AARCH64 SMP (similar to Cortex-A53).

Toolchain: GCC [version].

**Key Question**

For the configuration options configSTART_<Test_Name>_TESTS:

1. Should all tests pass when run simultaneously (all configSTART_*_TESTS set to 1)?

2. Or is it acceptable if tests pass only when run individually (testing one configSTART_*_TESTS at a time)?

aggarg, 2025-07-18:
It is hard to say that without looking at the failures. What is the value of `configRUN_MULTIPLE_PRIORITIES`? Can you try setting it to 0?

github-xiaodong, 2025-07-31:
Setting configUSE_CORE_AFFINITY=0 has no effect.

The kernel versions in RP2040 standard SMP demos and XCORE.AI_xClang RTOSDemo examples are somewhat outdated and not fully compatible with the current V11.2.0.

github-xiaodong, 2025-07-31:
Next, I will analyze why MessageBufferDemo and StreamBufferDemo fail during combined testing( set all configSTART_<Test_Name>TESTS to 1 ), but individual tests passing .